klack/marlin #100: Feature request: turn on firmware retractions

Buriaku, 2021-04-11:
Firmware retractions aren't activated in the source code at the moment.

They shouldn't interfere with anything unless G10, G11 or M209 is called.

This allows more scripts to be executed e.g. at layer change.
Some post-processing scripts require it e.g. the toolchange post-processing script implementing cura-style prime towers:
https://github.com/mkudzia84/toolchanger-pspp

klack, 2021-04-12:
Need to research the ramifications of turning this on, with regard to toolchanges, prusa slicer tool retraction (10mm default), and if it could affect filament change on screen or gcode.
 
[image]

Buriaku, 2021-04-12:
You're totally right. Hadn't thought about tool changes. You could set all default lengths to zero, I guess.

klack, 2021-04-16:
Moved to reddit

closed 2021-04-16 by klack