klack/marlin #103: Print does not start from SDCard

klack, 2021-04-16:
so, I start with a clarification:
- if I DO NOT preheat, what I wrote in the post happens: the print does not start and "finishing up ..." appears on the screen
the LCD screen freezes (if I "tap" nothing happens). but if I access the list of Gcode commands first, what appears up to before the block is this:

[image]

- if instead I do the preheat, then it happens that the print does not start the same but the temperature of the hotend slowly drops.
from the gcode command list, M118 repeats itself a few times
here is the picture:

[image]

Bru — Today at 12:30 AM
without prehet, when the "finishing up" screen appears, the LCD screen does not respond and I cannot abort the print and then go to the console.
with the preheat, on the other hand, the LCD works and this appears in the console, after the forced stop and the printer reset:
[image]

and .... ATTENTION, doing the tests to take the photos and send you the information, here, as if by magic, the printing started as always.
but I don't believe in magic.
something wrong.
same .gcode (then I attach it). I use IdeaMaker.

klack, 2021-04-16:
[LED_supp.gcode.txt](https://github.com/klack/marlin/files/6326854/LED_supp.gcode.txt)

klack, 2021-04-16:
Thanks for supplying all this Bru.  I do have a hunch on what it is.  

Can you please try:  

- Disable the "Monitor print status" command in Pronterface
[image]
- Connecting with pronterface and paste the output from there while it is frozen
- Issue M108 command from pronterface, up to 3 times

BruGiova, 2021-04-17:
the problem disappeared by itself.
it showed up for a dozen times (even after resetting the printer, switching off and switching on again, etc.) but as it appeared - then - it also disappeared.
i ran a few test prints (5 or 6) and everything seems to work again.
no intervention on my part (things like firmware reflash etc).
in case it shows up again, I will try this solution.
in any case: thanks for your interest.
Bru

BruGiova, 2021-05-01:
update:
it is the filament sensor that is causing the problem.
mine is an empirical deduction.

after many successful prints without problems, in one print I ran out of spool and the filament sensor intervened.
unfortunately the filament change routine did not start, but the damn word "finishing up ..." appeared. after the hotend went to homeX, the remaining filament was ejected and game over.
with the message on the display (impossible to intervene, the touch does not work) still present, the hotend temperature has started to drop.
all that's left is to turn the printer off and on again.
but the problem recurred randomly in subsequent prints (three attempts).
to find a solution, I disabled the sensor - with gcode M412 S0 - it is now always working all again without problems.
at the moment a file has been in print for over 20 hours.
let's see what happens....

klack, 2022-01-15:
Closing for now as filament sensor is a planned feature and not fully supported

closed 2022-01-15 by klack