klack/marlin #104: Crash when move after 2 G28 X0 in Duplication mode

murdock62, 2021-04-18:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
G28 ; home all
M605 S2 ; activate copy mode
G1 Z10 ; don't collide with bed
G28 X0 ; home x to activate copy mode
G1 X10 ; move both heads

It works like it should. But if you then do:
G28 X0
G1 X10

E1 will crash to the endstop.

**Expected behavior**
The two extruders must move simultaneously.

murdock62, 2021-04-18:
Seem to be fixed by modifying the g28.cpp file like this:

Before :
extruder_duplication_enabled = IDEX_saved_duplication_state;

After :
//extruder_duplication_enabled = IDEX_saved_duplication_state;

klack, 2022-01-15:
Need to reverify

commit ee199371 (2021-07-10): Added Murdock Bug Fix #104 & #33
commit ac68fcf0 (2021-07-14): Merge branch 'feature_marlin_bugfix_21992' of https://github.com/klack/marlin into feature_marlin_bugfix_21992

* 'feature_marlin_bugfix_21992' of https://github.com/klack/marlin:
  Comment platform.io
  Added Murdock Bug Fix #104 & #33