klack/marlin #111: Extruder movement problems when using the “Level Bed” Interface on the touchscreen (not consistent)

shrike64, 2021-04-30:
**Describe the bug**
When leveling bed using the touch panel, I noticed some occasional, intermittent problems with the movement of the extruders from one position to another.  

**To Reproduce**
To try to determine if there was some consistent pattern, I set up a sequence of positions to try, that I could repeat to see if the errors were random or if they had a pattern. I used the following (rather lengthy) sequence, starting with extruder 1 and alternating between the extruders several times:
1 – 2 – 4 – 3 – 5 – 4 – 2 – 1 – 3 – 5 – (swap extruder, now extruder 2)
3 – 4 – 1 – 2 – 5 – 2 – 1 – 4 – 3 – 5 – (swap extruder, now extruder 1)
3 – 2 – 4 – 1 – 5 – 2 – 4 – 3 – 1 – 5 – (swap extruder, now extruder 2)
1 – 3 – 2 – 4 – 5 – 4 – 3 – 2 – 1 – 5 – (swap extruder, now extruder 1)
4 – 2 – 3 – 1 – 5 – 3 – 1 – 4 – 2 – 5 – (swap extruder, now extruder 2)
4 – 2 – 3 – 1 – 5 – 3 – 1 – 4 – 2 – 5 – (swap extruder, now extruder 1)
[end of sequence]

For this test, I repeated this sequence three times, each time, starting the sequence after cycling the power on the printer. There was a similarity to the errors, but it was not consistent.  

1st run through the sequence: 
When at the 1st 5 of the 2nd row, when I pressed 2, extruder 2 moved from 5 to 2, but z did not drop to 0 (*1)
When at the 2nd 2 of the 5th row, when I pressed 5, extruder1 moved from 2 to 5, but z did not drop to 0 (*2)

2nd run through the sequence:
When at the 2nd 2 of the 2nd row, when I pressed 1, extruder 2 moved from 2 to 1, but z did not drop to 0 (*3)
When at the 1st 4 of the 6th row, when I pressed 2, extruder 2 moved from 4 to 2, but z did not drop to 0 (*4)

3rd run through the sequence:
When at the 2nd 2 of the 2nd row, when I pressed 1, extruder 2 moved from 2 to 1, but z did not drop to 0 (*5)
When at the 1st 1 of the 6th row, when I pressed 5, extruder 2 moved from 1 to 5, but z did not drop to 0 (*6)

During this test, other than the noted issues, all movements were normal and as expected.

(*1) – Command history:
M155 S2

M115
M503
M114
M155 S5
M220
M221

G28
T0
G0 F5000
M114
G90
G0 Z10 F1000
G0 X50 Y50 F5000
G0 Z0 F1000
M114
G90
G0 Z10 F1000
G0 X250 Y250 F5000
G0 Z0 F1000
M114
G90
G0 Z10 F1000
G0 X50 Y250 F5000
G0 Z0 F1000
M114
G90
G0 Z10 F1000
G0 X250 Y50 F5000
G0 Z0 F1000
M114
G90
G0 Z10 F1000
G0 X150 Y150 F5000
G0 Z0 F1000
M114
G90
G0 Z10 F1000
G0 X50 Y250 F5000
G0 Z0 F1000
M114
G90
G0 Z10 F1000
G0 X250 Y250 F5000
G0 Z0 F1000
M114
G90
G0 Z10 F1000
G0 X50 Y50 F5000
G0 Z0 F1000
M114
G90
G0 Z10 F1000
G0 X250 Y50 F5000
G0 Z0 F1000
M114
G90
G0 Z10 F1000
G0 X150 Y150 F5000
G0 Z0 F1000
M114
T1
M114
G90
G0 Z10 F1000
G0 X250 Y50 F5000
G0 Z0 F1000
M114
G90
G0 Z10 F1000
G0 X50 Y250 F5000
G0 Z0 F1000
M114
G90
G0 Z10 F1000
G0 X50 Y50 F5000
G0 Z0 F1000
M114
G90
G0 Z10 F1000
G0 X250 Y250 F5000
G0 Z0 F1000
M114
G90
G0 Z10 F1000
G0 X150 Y150 F5000
G0 Z0 F1000
M114
G90
G0 Z10 F1000
G0 X250 Y250 F5000
G0 Z0 F1000
M114
--END--

(*2) – Command history:
G0 X150 Y150 F5000
G0 Z0 F1000
M114
--END--

(*3) – Command history:
G0 X50 Y50 F5000
G0 Z0 F1000
M114
--END--

(*4) – Command history:
G0 X250 Y250 F5000
G0 Z0 F1000
M114
--END--

(*5) – Command history:
G0 X50 Y50 F5000
G0 Z0 F1000
M114
--END—

(*6) – Command history:
G0 X150 Y150 F5000
G0 Z0 F1000
M114
--END--

**Expected behavior**
Movement to the appropriate point, without the above noted anomalies

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - Marlin luxURI
 - 0.7.11

**Additional context**
From some other runs that were not as closely monitored:  Once the extruder rose, then dropped back in the same spot without moving to the position I had just chosen. On two other occasions, the extruder moved from the current position to the new position without moving up the usual 10 mm.

klack, 2021-04-30:
Very nice report, thank you!

klack, 2022-01-15:
Fixed this was a buffer issue on the TFT

closed 2022-01-15 by klack