klack/marlin #145: [BUG]  Exiting the Level bed screen can crash into front clips

dcbarry, 2022-03-29:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Under certain circumstances, upon exiting the “Level Bed” bed screen, the extruder returns to the off bed area, but crosses over the front of the print bed at a height of zero.  This can cause it to crash into and possible remove clips holding glass to the bed, a common placement.  

Does not happen for each level point —, and there is a difference between the E1 and E2 when exiting.

Note the extruder moves at nearly zero height along the front in the zone where clips are placed.  
(See video)

This does not happen when E2 is engaged and returning from the #1 location.  It takes a completely different return path.

### Bug Timeline

I just started testing 

### Expected behavior

Avoid the front edge —- possibly minim the E2 behavior

### Steps to Reproduce

To recreate:    Using E1 as selected extruder,

Enter the Level bed screen.   
Choose level spot 3  (front right)
Exit the screen (x)

### Version of Marlin Firmware

2.0.9.2 for luxuri 0.9.x

### Printer model

TL-03

### Add-ons

Stock Tenlog

### Bed Leveling

MBL Manual Bed Leveling

### Additional information & file uploads

Using your provided hex pre-builds

klack, 2022-05-28:
Issue has been fixed and will be in the next release

klack, 2022-10-10:
G0 Z5 F500 added before exit of level bed

closed 2022-10-10 by klack