klack/marlin #147: [BUG] (bug summary) Y axis does not home

enishi1357, 2022-05-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Y axis does not home. It moves a little bit down every time it home. 

### Expected behavior

I expect y-axis to move up to the max y axis position until it is detected by the y limit switch.

### Actual behavior

y-axis Moves a little bit down.

### Steps to Reproduce

1. Use brand new TL-D3 Pro that has the bmg extruder as stock and the TMC2209 drivers.
2. Use unedited compiled D3P_2209_Titan_BLTouch firmware
3. Install working Bltouch correctly
4. Use octoprint on a raspberry pi that's been used correctly on other machine . 
5. On web interface, Hit home for both x and y axis.

### Version of Marlin Firmware

D3P_2209_Titan_BLTouch

### Printer model

 TL-D3 Pro

### Electronics

Bltouch

### Bed Leveling

ABL Bilinear mesh

### Host Software

OctoPrint

enishi1357, 2022-05-28:
nvm I enabled invert y axis and now it works again.

enishi1357, 2022-05-28:
nvm the nvm. That didn't work.

enishi1357, 2022-05-28:
ok I had to enable y min not y max

klack, 2022-05-28:
Hi, make sure you are using the 2.0.9.2 branch.  This happens when you chose the incorrect Y endstop.

closed 2022-05-28 by klack