klack/marlin #150: [BUG] Move menu extrude button does not handle repeated press

jdulrich, 2022-05-30:
### Bug Description

Inside the Move menu, the buttons to extrude filament do not work as expected if pressed more than once.

### Bug Timeline

I noticed in build_ 20220422_2230 and touch_screen_2.0.9.tft but it's my first install

### Expected behavior

Press the button to extrude filament, and filament should extrude.

### Actual behavior

The first time I press the extrude button, filament extrudes correctly. Subsequent presses result in no activity. If I then change the distance setting from the default 10 to 1 and press extrude again, the filament retracts by 10.

### Steps to Reproduce

1. Turn on printer
2. Preheat nozzles in the Tools->Preheat menu
3. Navigate to the Move menu
4. Press the down arrow that should extrude filament - watch filament extrude
5. Press the same down arrow button again and observe no activity

### Version of Marlin Firmware

build_ 20220422_2230/D3P_2209_OpticalY_BLTouch

### Printer model

TL-D3

### Electronics

All stock with a BL Touch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

klack, 2022-05-30:
Thank you for the detailed report!

klack, 2022-10-10:
Fixed in 0.9.2

closed 2022-10-10 by klack