klack/marlin #164: filament advanced pause not working and crashes after approx 10 mins - due to screen sleeping

captaincreative, 2022-12-20:
### Bug Description

filament advanced pause not working and crashes after approx 10 mins - due to screen sleeping

### Bug Timeline

new bug 

### Expected behavior

pause for as long as needed to change filament, not really worth having filament pause if you need to watch it , idea to leave over night and swap filament in morning if required.

### Actual behavior

pause works but only for 10-15 mins approx..

### Steps to Reproduce

1) cut filament so it runs out
2) wait to see when screen goes off
3) awake screen and filament pause has crashed

### Version of Marlin Firmware

2.0.9.2

### Printer model

hictop d3

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

klack, 2022-12-22:
Removal of screen sleeping cause a bug where temp cannot be changed during sdprint, need to fix.

captaincreative, 2022-12-22:
yes i have noticed this aswell, i can not edit the temps during sd card print

klack, 2022-12-22:
Yes will take a look at this in the evening.

klack, 2023-01-06:
`tjc x3 202301052300` posted in gdrive for testing