klack/marlin #167: ABL probing off the bed

jladd1113, 2022-12-31:
### Bug Description

following the Luxuri wiki. added an official BL Touch. I measured and calculated my XY probe offset. They were about .5mm off so I'm confident that my offset settings are correct. M851 X-28 Y-48. The probe is mounted on the left front corner off E1. Used this mount:

https://cults3d.com/en/3d-model/tool/bltouch-bracket-mount-for-tenlog-tl-d3-pro

When I would execute G34, the probe would start about 20(ish)mm off the bed. I adjusted with M422. I believe the original was M422 S1 X15.00 Y151.00. I set X to 30.0 and it would probe on the bed.

Again I followed the wiki. G28, G34, G29 P0, G29 P1. Third probe is off the bed.

I then skipped the G34. Executed G28, G29 P0, G29 P1. 35th probe is off the bed.

trying to figure out why this is happening.

D3P_A4988_BLTouch

### Version of Marlin Firmware

Marlin 20020706_1711

### Printer model

Hictop D3 Hero

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

jladd1113, 2022-12-31:
I just realized that while running G34 and watching my terminal, that I am getting a probing failed and G34 aborted on the first iteration. I have never ran a G34 before, so I wasn't sure what the outcome was. I was blindly following the Wiki. I believe I should have 10 iterations at the end of a successful G34.

so it seems I have a different issue than originally stated.

jladd1113, 2022-12-31:
Figured out why G34 was failing

Adjusted M422 S2 as I realized it was set at 295mm which it is telling me it was out of range. I set it to 250.

Send: M422 S2
Recv: ?(X) out of bounds.

Recv:   M422 S1 X100.00 Y151.00
Recv:   M422 S2 X250.00 Y151.00

G34 ran successfully.

Recv: Z2-Z1=0.01
Recv: Target accuracy achieved.
Recv: Did 4 of 9
Recv: Accuracy: 0.01

So now it says E0 is sitting at the position below. 

Recv: echo:Active Extruder: 0
Recv: X:128.00 Y:199.00 Z:5.00 E:0.00 Count X:10240 Y:15920 Z:4000

Wiki states this:

**This procedure creates a map of the irregular shape of your printer bed and stores it for later use. Before doing this, it's a good idea to [tram your bed to the printer frame](https://github.com/klack/marlin/wiki/TramYourPrintBed) to make sure the bed is as level as possible.

From a terminal, enter G34. This synchronizes the two Z-axis motors to level the gantry to the bed.

Enter G29 P0 to clear out any stored bed mesh.

Enter G29 P1 to begin mesh generation. This will probe many points on your print bed and will take a while. Maybe 10 or 15 minutes.**

Send: G29 P0
Recv: Default storage slot 0 selected.
Recv: Mesh zeroed.
Recv: echo:Active Extruder: 0

Send: G29 P1
Recv: Default storage slot 0 selected.
Recv: Mesh invalidated. Probing mesh.

Recv: Probing mesh point 57/225.<---- this is where is goes off bed.

So it appears that G29 P1 starting out at X128 Y199 is the incorrect location to start?

My bed is 310x310. Stated printing volume is 300x300. The center of the bed would be 155x155. My probe when Z homing is at X183, Y199 which puts my nozzle at X211, Y247.

Should G29 be starting out at the middle of the bed? Should it be homed after G34 then the G29? Should it be starting at X183, Y199 where my probe hits during homing?  Are my M422 settings incorrect? It seems I am missing something.

klack, 2022-12-31:
Thank you for the reports.  I’ll be able to look more deeply after the
holidays