klack/marlin #168: [BUG] Can't compile anything with Build Flag "-DLCD_BTT_TFT"

gfacek, 2023-01-18:
### Bug Description

I tried to compile marlin-lux-marlin-2.0.9.2 with the following settings in luxuri.ini:
```
[env:D2P_OCTOPUS_Driver2209BTT_OpticalY_BLTouch_LCD_BTT_TFT]
platform           = ${common_stm32.platform}
extends            = stm32_variant
board              = marlin_BigTree_Octopus_v1
board_build.offset = 0x8000
build_flags       = ${stm32_variant.build_flags} -DMachineTLD2P -DOCTOPUS -DDriver2209BTT -DOpticalY -DBL_Touch -DLCD_BTT_TFT -DSTM32F446_5VX -DUSE_USB_HS_IN_FS
```
The compilation failed with the following errors:
`Marlin\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp:49:29: error: token "{" is not valid in preprocessor expressions
Marlin\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp:65:29: error: token "{" is not valid in preprocessor expressions
Marlin\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp:81:29: error: token "{" is not valid in preprocessor expressions`

### Expected behavior

I expected to compile the firmware.bin regarding the selected settings.

### Actual behavior

The firmware-file did not compile. Instead I got the following error:
`Marlin\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp:49:29: error: token "{" is not valid in preprocessor expressions
Marlin\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp:65:29: error: token "{" is not valid in preprocessor expressions
Marlin\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp:81:29: error: token "{" is not valid in preprocessor expressions`

### Steps to Reproduce

1. Add the following configuration in luxuri.ini:
```
[env:D2P_OCTOPUS_Driver2209BTT_OpticalY_BLTouch_LCD_BTT_TFT]
platform           = ${common_stm32.platform}
extends            = stm32_variant
board              = marlin_BigTree_Octopus_v1
board_build.offset = 0x8000
build_flags       = ${stm32_variant.build_flags} -DMachineTLD2P -DOCTOPUS -DDriver2209BTT -DOpticalY -DBL_Touch -DLCD_BTT_TFT -DSTM32F446_5VX -DUSE_USB_HS_IN_FS
```
2. Select the configuration in platformio.ini as default env:
`default_envs = D2P_OCTOPUS_Driver2209BTT_OpticalY_BLTouch_LCD_BTT_TFT`
3. Compile

### Version of Marlin Firmware

marlin-lux-2.0.9.2

### Printer model

Tenlog Hands 2

### Electronics

BTT Octopus Board and BTT TFT35

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

OctoPrint

gfacek, 2023-01-31:
Found a solution:

in Marlin/src/pins/stm32f4/pins_BTT_OCTOPUS_V1_common.h there is this part:
`#if HAS_MARLINUI_U8GLIB
  #ifndef BOARD_ST7920_DELAY_1
    #define BOARD_ST7920_DELAY_1   DELAY_NS(120)  // DELAY_NS(96)
  #endif
  #ifndef BOARD_ST7920_DELAY_2
    #define BOARD_ST7920_DELAY_2   DELAY_NS(80)   // DELAY_NS(48)
  #endif
  #ifndef BOARD_ST7920_DELAY_3
    #define BOARD_ST7920_DELAY_3   DELAY_NS(580)  // DELAY_NS(600)
  #endif
#endif`

I had to change it to:
`#if HAS_MARLINUI_U8GLIB
  #ifndef BOARD_ST7920_DELAY_1
    #define BOARD_ST7920_DELAY_1   120  // DELAY_NS(96)
  #endif
  #ifndef BOARD_ST7920_DELAY_2
    #define BOARD_ST7920_DELAY_2   80   // DELAY_NS(48)
  #endif
  #ifndef BOARD_ST7920_DELAY_3
    #define BOARD_ST7920_DELAY_3   580  // DELAY_NS(600)
  #endif
#endif`

klack, 2023-01-31:
Great work!  This one has stumped us.  I'm keeping this bug open until it can be implemented in our new Marlin 2.1x build.  Please join us in Discord if you haven't already:
https://discord.gg/w8gMzQq8Bp

gfacek, 2023-02-01:
Glad I could help :)
And thanks for the Discord invite!